micro-ROS/rmw_microxrcedds #12: runtime error on ROS 2 Crystal

iluetkeb, 2019-01-14:
I set RMW_IMPLEMENTATION=rmw_microxrcedds, then on `ros2 run examples_rclcpp_minimal_publisher publisher_lambda` I get the following error:

`'failed to resolve symbol 'rmw_get_serialization_format' in shared library '/home/lui3si/ros2_ws/install/lib/librmw_microxrcedds.so', at /tmp/binarydeb/ros-crystal-rmw-implementation-0.6.1/src/functions.cpp:162'`

Moreover, I also get the error 
`'type support not from this implementation, at /home/lui3si/ros2_ws/src/rmw-microxrcedds/rmw_microxrcedds_c/src/rmw_publisher.c:66'`

This is on Ubuntu 18.04.

Both rmw_microxrcedds and rosidl_typesupport_microxrcedds are using the "feature/Crystal_update" branch, because master did not compile.

iluetkeb, 2019-01-14:
One thing could be that I used the thirdparty fastrtps, whereas ros crystal uses the fastrtps from /opt/ros/...

BorjaOuterelo, 2019-01-14:
Solved in c9d9ad49defb1732f98dcb6cd2df9ed84bce9d73 and 146c19d2e71af37a130fea2965d08643ea1d9454

Those commits:
-Add missing crystal new API.
-Add the usage of Poco library to choose type support.

Can you try with the branch feature/Crystal_update?
You should be able to run either C or CPP examples:
-ros2 run int32_publisher_c  int32_publisher_c
-ros2 run examples_rclcpp_minimal_publisher publisher_lambda

iluetkeb, 2019-01-15:
With the recent updates, I now get
lui3si@crystal-devel:~/ros2_ws$ ros2 run examples_rclcpp_minimal_publisher publisher_lambda
UDP mode => ip: 127.0.0.1 - port: 8888
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  could not create publisher: type support not from this implementation, at /home/lui3si/ros2_ws/src/rmw-microxrcedds/rmw_microxrcedds_c/src/rmw_publisher.c:64, at /tmp/binarydeb/ros-crystal-rcl-0.6.3/src/rcl/publisher.c:173

If it works on your box, what could be the difference? I'm building with -DTHIRDPARTY, but using rcl and rmw from /opt/ros/crystal/. Are you guys doing the same?

BorjaOuterelo, 2019-01-15:
We are not building with -DTHIRDPARTY, we are using 
`colcon build --cmake-args -DBUILD_SHARED_LIBS=ON`

I am using the following docker image https://github.com/microROS/docker/blob/feature/Crystal_update/Linux/development/micro-ROS_development/Dockerfile as build environment.

And these are the micro-ROS related repositories added to the Crystal released ones:
```
# Micro XRCE-DDS
  eProsima/Micro-CDR:
    type: git
    url: https://github.com/eProsima/micro-CDR.git
    # Change2Tag
    version: master
  eProsima/Micro-XRCE-DDS-Client:
    type: git
    url: https://github.com/eProsima/Micro-XRCE-DDS-Client.git
    # Change2Tag
    version: master

# MicroROS
  uros/rclc:
    type: git
    url: https://github.com/microROS/rclc.git
    # Change2Tag
    version: feature/Crystal_update
  uros/micro-ROS-demo:
    type: git
    url: https://github.com/microROS/micro-ROS-demos.git
    # Change2Tag
    version: feature/DemoFix
  uros/rmw_microxrcedds:
    type: git
    url: https://github.com/microROS/rmw-microxrcedds.git
    # Change2Tag
    version: feature/Crystal_update
  uros/rosidl_typesupport_microxrcedds:
    type: git
    url: https://github.com/microROS/rosidl_typesupport_microxrcedds.git
    # Change2Tag
    version: feature/Crystal_update
```

iluetkeb, 2019-01-15:
Can you post your full repository YAML spec? When combining the above with the one in the micro-ROS install docs, I get compile errors. Most likely, you have updated version in other repositories as well.

btw, I have tried combining the above with ROS 2 installed from packages, to no avail. So now the next try is compile everything from scratch.

BorjaOuterelo, 2019-01-16:
@iluetkeb Did you manage to compile it?

I use the Crystal list:

```
repositories:
  ament/ament_cmake:
    type: git
    url: https://github.com/ament/ament_cmake.git
    version: 0.6.0
  ament/ament_index:
    type: git
    url: https://github.com/ament/ament_index.git
    version: 0.5.1
  ament/ament_lint:
    type: git
    url: https://github.com/ament/ament_lint.git
    version: 0.6.2
  ament/ament_package:
    type: git
    url: https://github.com/ament/ament_package.git
    version: 0.6.0
  ament/googletest:
    type: git
    url: https://github.com/ament/googletest.git
    version: c8b6f2be853002b9a0abfb30ea537402383bf1f4
  ament/osrf_pycommon:
    type: git
    url: https://github.com/osrf/osrf_pycommon.git
    version: 0.1.6
  ament/uncrustify_vendor:
    type: git
    url: https://github.com/ament/uncrustify_vendor.git
    version: 1.0.0
  eProsima/Fast-CDR:
    type: git
    url: https://github.com/eProsima/Fast-CDR.git
    version: v1.0.8
  eProsima/Fast-RTPS:
    type: git
    url: https://github.com/eProsima/Fast-RTPS.git
    version: b48ce9d2fba6fc94e756da01c58b72f2ad848238
  osrf/osrf_testing_tools_cpp:
    type: git
    url: https://github.com/osrf/osrf_testing_tools_cpp.git
    version: 1.1.0
  ros/class_loader:
    type: git
    url: https://github.com/ros/class_loader.git
    version: 1.2.0
  ros/pluginlib:
    type: git
    url: https://github.com/ros/pluginlib.git
    version: 2.2.1
  ros/resource_retriever:
    type: git
    url: https://github.com/ros/resource_retriever.git
    version: 2.1.0
  ros/ros_environment:
    type: git
    url: https://github.com/ros/ros_environment.git
    version: 2.2.0
  ros/urdfdom_headers:
    type: git
    url: https://github.com/ros/urdfdom_headers.git
    version: 1.0.2
  ros-perception/laser_geometry:
    type: git
    url: https://github.com/ros-perception/laser_geometry.git
    version: 2.0.0
[... 261 lines cut ...]
# MicroROS
  uros/rclc:
    type: git
    url: https://github.com/microROS/rclc.git
    # Change2Tag
    version: feature/Crystal_update
  uros/micro-ROS-demo:
    type: git
    url: https://github.com/microROS/micro-ROS-demos.git
    # Change2Tag
    version: feature/DemoFix
  uros/rmw_microxrcedds:
    type: git
    url: https://github.com/microROS/rmw-microxrcedds.git
    # Change2Tag
    version: feature/Crystal_update
  uros/rosidl_typesupport_microxrcedds:
    type: git
    url: https://github.com/microROS/rosidl_typesupport_microxrcedds.git
    # Change2Tag
    version: feature/Crystal_update
```

The ones present in https://raw.githubusercontent.com/microROS/micro-ROS-doc/feature/RepoListUpdate/Installation/repos/client_minimum.repos are a selected subset of these lists, containing just the required content for the Client.
And there is an analogous one for the Agent.

BorjaOuterelo, 2019-01-16:
> btw, I have tried combining the above with ROS 2 installed from packages, to no avail. So now the next try is compile everything from scratch.

I always compile from sources. I may need to have a look that issue with the ROS 2 installed from packages.

pablogs9, 2021-02-08:
Closing due Crystal in EOL: https://index.ros.org/doc/ros2/Releases/

closed 2021-02-08 by pablogs9