micro-ROS/rmw_microxrcedds #29: Build fail on micro-ROS Client for NuttX (error: unknown type name 'SampleIdentity')

jfm92, 2020-01-27:
From the commit (b5eea0997a3884263d066f9413c9c049ce9334bb)of rmw_microxrcedds the build process fails with the micro-ROS build system.

It retuns the next error:
```bash
/uros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/./types.h:81:3: error: unknown type name 'SampleIdentity'
   SampleIdentity sample_id[RMW_UXRCE_MAX_HISTORY];
   ^~~~~~~~~~~~~~
make[6]: *** [CMakeFiles/rmw_microxrcedds.dir/src/rmw_publisher.c.obj] Error 1
make[5]: *** [CMakeFiles/rmw_microxrcedds.dir/all] Error 2
make[4]: *** [all] Error 2
---
Failed   <<< rmw_microxrcedds	[ Exited with code 2 ]

Summary: 17 packages finished [8.51s]
  1 package failed: rmw_microxrcedds
  3 packages had stderr output: libyaml_vendor rcutils rmw_microxrcedds
  31 packages not processed
Makefile:34: recipe for target 'colcon_compile' failed
make[3]: *** [colcon_compile] Error 2
make[3]: se sale del directorio '/home/juan/demo_ingo/prueba_bug/uros_ws/firmware/apps/uros'
Makefile:106: recipe for target 'uros_context' failed
make[2]: *** [uros_context] Error 2
make[2]: se sale del directorio '/home/juan/demo_ingo/prueba_bug/uros_ws/firmware/apps'
Makefile:179: recipe for target 'context_serialize' failed
make[1]: *** [context_serialize] Error 2
make[1]: se sale del directorio '/home/juan/demo_ingo/prueba_bug/uros_ws/firmware/apps'
tools/Makefile.unix:345: recipe for target 'context' failed
make: *** [context] Error 2
```

I'm using the next repos:
- Micro-XRCE-DDS Client (b5eea0997a3884263d066f9413c9c049ce9334bb)
- Micro-ROS-Build (b178c62353f800cc8b9c5160c46a8d24d56cc2e0) with the modified repos.
- apps (8e1dd60ac8f86f1d0ce16744a92837b764a9a9f3)
- Nuttx (5f624f7988ebaf16ad661688696d9cc6a8ed58b2)

iluetkeb, 2020-01-27:
Which command did you use to perform the build?

jfm92, 2020-01-27:
@iluetkeb, I've used: ``ros2 run micro_ros_setup build_firmware.sh``.
I've updated the specific commit of each repo. The issue only appears on that commit and afterward, with (8ce9eba38c945e62bdf1c93cd449f2af9ac80fbb) works fine.

iluetkeb, 2020-01-27:
As far as I can tell, the `SampleIdentity` type is not defined in micro-xrce-dds-client. Where is it defined?

iluetkeb, 2020-02-04:
Is this still an issue?

pablogs9, 2020-02-17:
`SampleIdentity` is a definition used by Micro-XRCE-DDS-Client implementation of services. Services are included in dashing version in this commit: https://github.com/eProsima/Micro-XRCE-DDS-Client/commit/fd52a2505f5efc0ce7071a7780035c0fc82076bf

Please @jfm92 close if resolved.

pablogs9, 2020-11-16:
Closing bc outdated

closed 2020-11-16 by pablogs9