micro-ROS/rmw_microxrcedds #37: Repo cleaning

julibert, 2020-01-29:
The following is a list of suggested actions to clean-up this repo:

- [ ] rebase branches: [remove_unused_variable], [remove_duplicate_structs], [best_effort] and [6lowpan] over [dashing],
- [ ] change #36 target branch to [dashing],
- [ ] remove [master] branch.

[dashing]: https://github.com/micro-ROS/rmw-microxrcedds/tree/dashing
[master]: https://github.com/micro-ROS/rmw-microxrcedds/tree/master
[remove_unused_variable]: https://github.com/micro-ROS/rmw-microxrcedds/tree/feature/remove_unused_variable
[remove_duplicate_structs]: https://github.com/micro-ROS/rmw-microxrcedds/tree/feature/remove_duplicated_structs
[best_effort]: https://github.com/micro-ROS/rmw-microxrcedds/tree/test/best_effort
[6lowpan]: https://github.com/micro-ROS/rmw-microxrcedds/tree/feature/6lowpan

iluetkeb, 2020-01-29:
I changed the base on #36, but it also needs a re-base most likely.

- [ ]  rebase #36

julibert, 2020-01-29:
Thanks @iluetkeb. Rebased!

closed 2021-02-08 by pablogs9