micro-ROS/rmw_microxrcedds #39: sequence deserialize error

Aiven92, 2020-02-02:
Some error with sequence deserialization.
message allocated before call `rclc_spin_node`, used custom `rclc_spin_allocatedmsg_node_once(rclc_node_t * already_allocated_node, void* msg, int64_t timeout_ms)`

Tested on 
 - joint_trajectory_point.
 - custom_msg( float64[] )

```
any_complex_message__cdr_deserialize(...){
  {
    uint32_t size;
    const size_t capacity = ros_message->array.capacity;
    rv = ucdr_deserialize_sequence_double(cdr, ros_message->array.data, capacity, &size);

    //size of sequence data detected as correct

    ros_message->array.size = size;
  }
}
```
Probably error in `ucdr_buffer_alignment` or  `ucdr_check_final_buffer_behavior`

#26 looks similar, but I'm think there is problem with message allocation.

FranFin, 2020-07-20:
@Aiven92 Hi! Thank you for your interest in micro-ROS. Notice that we’ll discuss the recent contributions and developments in the next [Embedded WG meeting](https://discourse.ros.org/t/ros-2-embedded-wg-meetings/15460). You’re invited to bring your micro-ROS related issue to this platform. We’ll be very happy to have you join us and discuss your questions with the community.

pablogs9, 2020-11-16:
Solved here: https://github.com/micro-ROS/rosidl_typesupport_microxrcedds/pull/25

closed 2020-11-16 by pablogs9