micro-ROS/rmw_microxrcedds #44: Problem with finding SampleIdentity type in compilation

Sdehnavi, 2020-02-13:
Hi,
Thanks for your contribution on micro-ROS.
I am trying to run an example on micro-ROS, but there is a problem in <types.h>. The problem is that it can not find SampleIdentity.
Here is the error: types.h:80:3: error: unknown type name ‘SampleIdentity’
Where is the header file containing SampleIdentity so tha I can include it in types.h?
However, it is still my question why should we use SampleIdentity in the client part of the library while SampleIdentity is a class of FastRTPS and in the client part of micro-ROS we don't use FastRTPS.

Thanks to your response in advance.

pablogs9, 2020-03-10:
Is this issue currently active?

pablogs9, 2020-11-16:
Closing bc inactive

closed 2020-11-16 by pablogs9