micro-ROS/rmw_microxrcedds #47: ‘UXR_REQUESTER_ID’ undeclared in rmw_node.c

Sdehnavi, 2020-02-14:
Hi,

I get the following error in rmw_node.c
rmw_client.c:110:69: error: ‘UXR_REQUESTER_ID’ undeclared (first use in this function); did you mean ‘UXR_REQUEST_NONE’?
     custom_client->client_id = uxr_object_id(custom_node->id_gen++, UXR_REQUESTER_ID);

Would you please tell me where UXR_REQUESTER_ID has been defined?
Thanks!

pablogs9, 2020-02-17:
Hello, can you detail which is your Micro-ROS setup?

It seems that you have a version of Micro-XRCE-DDS-Client without services, because the value you are refering is related to services implementation. They are implemented in this commit: https://github.com/eProsima/Micro-XRCE-DDS-Client/commit/fd52a2505f5efc0ce7071a7780035c0fc82076bf

Sdehnavi, 2020-02-17:
Thanks @pablogs9 
I changed to the crystal version and now it has been solved. But I have another issue which is related to type_support. I will create a new issue for it.

closed 2020-02-17 by Sdehnavi