micro-ROS/rmw_microxrcedds #50: Issues creating micro XRCE-DDS entities - publisher example app

krm-mir, 2020-02-17:
I'm having trouble setting up a session between my **olimex-stm32-e407** board running **NuttX** and the publisher example with microROS and mu pc running the microROS agent. Am using udp only in my project. Right now I'm just trying to understand the process when the board and agent communicates via microXRCEDDS.

**Output from nuttshell:**
`UDP mode => ip: 192.168.12.254 - port: 8888
Node initialization error: Issues creating micro XRCE-DDS entities, at /home/krm/olimex_stm32_test_s1
or() should be called after error handling to avoid this.`

**Output from agent:**
`Enter command: udp --port= 8888
Enter 'q' for exit
[1581950563.244114] info     | UDPServerLinux.cpp | init                     | running...             | port: 8888
[1581950571.452431] info     | Root.cpp           | create_client            | create                 | client_key: 0x24189DCF, session_id: 0x81
[1581950571.452525] info     | UDPServerBase.cpp  | on_create_client         | session established    | client_key: 0x24189DCF, address: 192.168.12.251:260
[1581950571.465038] info     | Root.cpp           | delete_client            | delete                 | client_key: 0x24189DCF
[1581950571.465094] info     | UDPServerBase.cpp  | on_delete_client         | session closed         | client_key: 0x24189DCF, address: 192.168.12.251:260
`

it seems like i get a nullpointer when rmw initializes a publisher node for the '/rosout' topic when i debug the olimex - at "mcu_ws/ros2/rcl/rcl/src/rcl/publisher.c, line 170" i get an error that goes to 'fail'.

I have picked up some packets with wireshark seeing that the pc joins RTPS with multicast, but never the olimex. Please ask if i should post this as well.

Any suggestions?
Also, i was unsure where to post this issue, so hope i chose correctly, and it might be solved in #42 , not sure.
Thanks.

**I am running the following:**
- Ubuntu 18.04 on pc
- ROS2 Dashing with micro-ros-build dashing branch installed
- I used the micro_ros_setup scripts to create both the firmware_ws and agent_ws
- pc and olimex are connected via router acting as dhcp (NuttX is configured to ask dhcp for an ip)

**I've made following changes from the original repos:**
- Changed following in the uROS_agent:
```
 git diff origin/dashing 
diff --git a/micro_ros_agent/src/main.cpp b/micro_ros_agent/src/main.cpp
index 7fc0cf3..47067b3 100644
--- a/micro_ros_agent/src/main.cpp
+++ b/micro_ros_agent/src/main.cpp
@@ -35,10 +35,8 @@ int main(int argc, char** argv)
     app.get_formatter()->column_width(42);
 
     /* CLI subcommands. */
-    eprosima::uxr::cli::UDPv4Subcommand udpv4_subcommand(app);
-    eprosima::uxr::cli::UDPv6Subcommand udpv6_subcommand(app);
-    eprosima::uxr::cli::TCPv4Subcommand tcpv4_subcommand(app);
-    eprosima::uxr::cli::TCPv6Subcommand tcpv6_subcommand(app);
+    eprosima::uxr::cli::UDPSubcommand udp_subcommand(app);
+    eprosima::uxr::cli::TCPSubcommand tcp_subcommand(app);
 #ifndef _WIN32
     eprosima::uxr::cli::SerialSubcommand serial_subcommand(app);
     eprosima::uxr::cli::PseudoSerialSubcommand pseudo_serial_subcommand(app);
```
- Changed the apps repo being pulled to my own copied version (but the publisher example hasn't changed)
- Edited the CMakeList in the firmware/mcu_ws for uros to make the publisher target my pc's ip instead of localhost:
```
git diff origin/dashing 
diff --git a/rmw_microxrcedds_c/CMakeLists.txt b/rmw_microxrcedds_c/CMakeLists.txt
index 6f2726f..af1fc03 100644
--- a/rmw_microxrcedds_c/CMakeLists.txt
+++ b/rmw_microxrcedds_c/CMakeLists.txt
@@ -60,7 +60,7 @@ if(${RMW_UXRCE_TRANSPORT} STREQUAL "serial")
   set(RMW_UXRCE_DEFAULT_SERIAL_DEVICE "/dev/ttyAMA0" CACHE STRING "Sets the agent serial port.")
 elseif(${RMW_UXRCE_TRANSPORT} STREQUAL "udp")
   set(MICRO_XRCEDDS_UDP ON)
-  set(RMW_UXRCE_DEFAULT_UDP_IP "127.0.0.1" CACHE STRING "Sets the agent IP address.")
+  set(RMW_UXRCE_DEFAULT_UDP_IP "192.168.12.254" CACHE STRING "Sets the agent IP address.")
   set(RMW_UXRCE_DEFAULT_UDP_PORT "8888" CACHE STRING "Sets the agent IP port.")
 elseif(${RMW_UXRCE_TRANSPORT} STREQUAL "custom")
   set(MICRO_XRCEDDS_CUSTOM ON)
```

pablogs9, 2020-02-18:
Please, try to run the agent with the -v6 flag in order to increase verbosity and update the agent log. 

Please comment your RMW and Micro-XRCE-DDS configuration, in nuttx you can find them in `apps/uros/rmw_config.meta` and in `firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/client.config`

Regarding the chenges you have, maybe it is a good idea to replicate an out of the box example from the micro-ROS-setup repo: (https://github.com/micro-ROS/micro-ros-build/tree/dashing/micro_ros_setup). The Nuttx over UDP should work as it is.

krm-mir, 2020-02-18:
**mcu_ws/eProsima/Micro-XRCE-DDS-Client/client.config:**
```
PROFILE_DISCOVERY=TRUE
PROFILE_UDP_TRANSPORT=TRUE
PROFILE_TCP_TRANSPORT=FALSE
PROFILE_SERIAL_TRANSPORT=TRUE

CONFIG_MAX_OUTPUT_BEST_EFFORT_STREAMS=1
CONFIG_MAX_OUTPUT_RELIABLE_STREAMS=1
CONFIG_MAX_INPUT_BEST_EFFORT_STREAMS=1
CONFIG_MAX_INPUT_RELIABLE_STREAMS=1

CONFIG_MAX_SESSION_CONNECTION_ATTEMPTS=5
CONFIG_MIN_SESSION_CONNECTION_INTERVAL=500
CONFIG_MIN_HEARTBEAT_TIME_INTERVAL=1

CONFIG_BIG_ENDIANNESS=FALSE

CONFIG_UDP_TRANSPORT_MTU=512
CONFIG_TCP_TRANSPORT_MTU=512
CONFIG_SERIAL_TRANSPORT_MTU=512
```

**Agent log:**
```
Enter command: udp -v6 --port= 8888
Enter 'q' for exit
[1582010689.303195] info     | UDPServerLinux.cpp | init                     | running...             | port: 8888
[1582010689.303622] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1582010711.762080] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 5E 2B 2B F7 81 00 FC 01
[1582010711.762341] info     | Root.cpp           | create_client            | create                 | client_key: 0x5E2B2BF7, session_id: 0x81
[1582010711.762423] info     | UDPServerBase.cpp  | on_create_client         | session established    | client_key: 0x5E2B2BF7, address: 192.168.12.253:260
[1582010711.762588] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x5E2B2BF7, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1582010711.763335] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5E2B2BF7, len: 108, data: 
0000: 81 80 00 00 01 05 62 00 00 0A 00 01 01 02 00 00 54 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 69 6E 74 33 32 5F 70 75 62 6C 69 73 68 65
0040: 72 5F 72 63 6C 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63 69 70 61 6E 74
0060: 3E 3C 2F 64 64 73 3E 00 00 00 00 00
[1582010711.770579] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x5E2B2BF7, participant_id: 0x000(1)
[1582010711.770777] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x5E2B2BF7, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1582010711.770825] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x5E2B2BF7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1582010711.771447] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5E2B2BF7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1582010711.772026] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5E2B2BF7, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1582010711.772283] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x5E2B2BF7, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1582010711.773388] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5E2B2BF7, len: 12, data: 
0000: 81 00 00 00 03 01 04 00 00 02 FF FE
[1582010711.773740] debug    | UDPServerLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5E2B2BF7, len: 12, data: 
0000: 81 00 00 00 03 01 04 00 00 02 FF FE
[1582010711.775128] info     | Root.cpp           | delete_client            | delete                 | client_key: 0x5E2B2BF7
[1582010711.775179] info     | UDPServerBase.cpp  | on_delete_client         | session closed         | client_key: 0x5E2B2BF7, address: 192.168.12.253:260
[1582010711.775322] debug    | UDPServerLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x00000000, len: 14, data: 
0000: 81 00 00 00 05 01 06 00 00 02 FF FE 00 00
```

I can't find rmw_config.meta, but i did change the mcu_ws/install/rmw_microxrcedds/include/rmw_microxrcedds_c/config.h @RMW_UXRCE_MAX_PUBLISHERS_X_NODE as per #42 (if i understood that correctly)
```
#ifndef RMW_MICROXRCEDDS_CONFIG_H
#define RMW_MICROXRCEDDS_CONFIG_H

#include <uxr/client/config.h>

#define MICRO_XRCEDDS_UDP
/* #undef MICRO_XRCEDDS_SERIAL */
/* #undef MICRO_XRCEDDS_CUSTOM */
/* #undef MICRO_XRCEDDS_USE_REFS */
#define MICRO_XRCEDDS_USE_XML

#ifdef MICRO_XRCEDDS_UDP
    #define RMW_UXRCE_MAX_TRANSPORT_MTU UXR_CONFIG_UDP_TRANSPORT_MTU
    // #define RMW_UXRCE_DEFAULT_UDP_IP "192.168.12.254"
    // #define RMW_UXRCE_DEFAULT_UDP_PORT 8888
    #define RMW_UXRCE_DEFAULT_UDP_IP "192.168.12.254"
    #define RMW_UXRCE_DEFAULT_UDP_PORT "8888"
#elif defined(MICRO_XRCEDDS_SERIAL)
    #define RMW_UXRCE_MAX_TRANSPORT_MTU UXR_CONFIG_UDP_TRANSPORT_MTU
    #define RMW_UXRCE_DEFAULT_SERIAL_DEVICE ""
#elif defined(MICRO_XRCEDDS_CUSTOM)
    #define RMW_UXRCE_MAX_TRANSPORT_MTU UXR_CONFIG_SERIAL_TRANSPORT_MTU
#endif

#define RMW_UXRCE_MAX_HISTORY 1
#define RMW_UXRCE_MAX_BUFFER_SIZE (RMW_UXRCE_MAX_TRANSPORT_MTU * RMW_UXRCE_MAX_HISTORY)

#define RMW_UXRCE_MAX_NODES 1
// #define RMW_UXRCE_MAX_PUBLISHERS_X_NODE 2
#define RMW_UXRCE_MAX_PUBLISHERS_X_NODE 1
#define RMW_UXRCE_MAX_SUBSCRIPTIONS_X_NODE 1
#define RMW_UXRCE_MAX_SERVICES_X_NODE 1
#define RMW_UXRCE_MAX_CLIENTS_X_NODE 1

#define RMW_UXRCE_NODE_NAME_MAX_LENGTH 128
#define RMW_UXRCE_TOPIC_NAME_MAX_LENGTH 100
#define RMW_UXRCE_TYPE_NAME_MAX_LENGTH 128
#define RMW_UXRCE_XML_BUFFER_LENGTH 400
#define RMW_UXRCE_REF_BUFFER_LENGTH 100

#endif // RMW_MICROXRCEDDS_CONFIG_H
```

pablogs9, 2020-02-18:
You should not modify files inside install and build folders because they are overwrite on each build. Please try to increase your publishers using micro-ROS config. In Nuttx you can find doing `make menuconfig` in Nuttx folder and under apps you would find the micro-ROS parameters.

krm-mir, 2020-02-18:
I replicated with a fresh install of micro_ros_setup - this works as it should.
I guess i have either changed to much in my old install or missed to pull some commits.
Thanks

closed 2020-02-18 by krm-mir