micro-ROS/rmw_microxrcedds #116: Support session time sync

amfern, 2021-03-24:
I am looking to sync time between the agent and the client, the eProsima client has a handy function for doing just that `uxr_sync_session` but its not exposed through the `rmw-microxrcedds` api. Is there a reason its not supported? 
I am willing to open PR for this

## Issue template

- Hardware description: 
stm32 micro controller
- Installation type: 
ros2arduino

pablogs9, 2021-03-24:
Hello, this feature is not implemented yet in micro-ROS but there no problem in implementing it.

What would you expect from it? Is it enough for you to retrieve a time difference between the client clock and the agent clock (POSIX time)? Or maybe it is better to obtain directly the POSIX time of the agent?

amfern, 2021-03-24:
I need the POSIX time, to fill the `stamp` part of the ros2 `std_msgs/header`

Acuadros95, 2021-03-25:
Hello, I have opened a PR for this issue: #117 

The implemented function `rmw_uros_sync_session` returns the micro-ROS Agent POSIX time in nanoseconds, does this work for you?

pablogs9, 2021-03-26:
Merged, #117, I'm closing to keep issue track clean. Please @amfern reopen if you need any other feature related with timing or any changes in this implementation, we are open to discuss.

Thanks, @Acuadros95.

closed 2021-03-26 by pablogs9

amfern, 2021-03-26:
Thanks, that works for me.