micro-ROS/rmw_microxrcedds #203: rmw_uros_ping_agent can cause exception when micro-ROS has not been initialized (MULTITHREAD Feature on)

wonjae-song, 2021-11-24:
- Hardware description: N/A
- RTOS: freeRTOS
- Installation type: micro_ros_setup (generate_lib)
- Version or commit hash: galactic

#### Steps to reproduce the issue
option : -DUCLIENT_PROFILE_MULTITHREAD=On -DPLATFORM_NAME_FREERTOS
step 1> initialize custom transport
step 2> call "rmw_uros_ping_agent" to check if microros agent is available
step 3> initialize micro-ros (call rclc_support_init, rclc_node_init_default and so on)

#### Expected behavior
wait for agent to be ready at step 2
so I can initialize micro-ros after success to ping

#### Actual behavior
exception is triggered at step2.
`UXR_LOCK(&session_memory.mutex); //<--- session_memory.mutex seems not to be initialized`

#### Additional information
according to API description, rmw_uros_ping_agent can be used even when the micro-ROS context has not yet been initialized, which is useful to test the connection before trying to connect to the Agent

pablogs9, 2021-11-24:
Working on this

pablogs9, 2021-11-24:
Related: https://github.com/micro-ROS/rmw_microxrcedds/pull/205

closed 2021-12-17 by pablogs9