micro-ROS/rmw_microxrcedds #216: Handle's typesupport identifier (rosidl_typesupport_c) is not supported by this library [galactic]

russkel, 2021-12-02:
**Describe the bug**

I have followed the tutorial https://micro.ros.org/docs/tutorials/core/first_application_linux/ and when it comes time to run the MicroROS agent and the ping_pong demo, I am getting this error:

```
[russ@sherlock microros_ws]$ ros2 run micro_ros_demos_rclc ping_pong

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'Handle's typesupport identifier (rosidl_typesupport_c) is not supported by this library, at /tmp/binarydeb/ros-galactic-rosidl-typesupport-c-1.2.1/src/type_support_dispatch.hpp:113'

with this new error message:

  'Undefined type support, at /home/russ/Devel/ros/microros_ws/src/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_publisher.c:127'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1638442499.958701508] [rclc]: [rclc_publisher_init_best_effort] Error in rcl_publisher_init: Undefined type support, at /home/russ/Devel/ros/microros_ws/src/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_publisher.c:127, at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/publisher.c:114

Failed status on line 89: 1. Aborting.
[russ@sherlock microros_ws]$ 
```

**To Reproduce**
Steps to reproduce the behaviour:
1. Follow tutorial https://micro.ros.org/docs/tutorials/core/first_application_linux/
2. Run MicroROS agent
3. Attempt to run `ping_pong` demo.
4. See error

**Expected behaviour**
Pings and pongs ;)

**System information (please complete the following information):**
 - OS: Ubuntu 20.04 (running via chroot on Manjaro Linux)
 - ROS 2 Galactic
 - Version - Fresh checkout of all repos. All ROS2 packages up to date as per published packages.

pablogs9, 2021-12-02:
I have replicated the tutorial and it works as expected: https://asciinema.org/a/4dhLRFdPu0VusGKeQoWDwKEu3

Can you check that you are using the commands correctly?

russkel, 2021-12-02:
> I have replicated the tutorial and it works as expected: https://asciinema.org/a/4dhLRFdPu0VusGKeQoWDwKEu3
> 
> Can you check that you are using the commands correctly?

Weird. I will try it again.

Is this error:
```
'Handle's typesupport identifier (rosidl_typesupport_c) is not supported by this library, at /tmp/binarydeb/ros-galactic-rosidl-typesupport-c-1.2.1/src/type_support_dispatch.hpp:113'
```
Because the typesupport should be `rosidl_typesupport_microxrcedds_c` instead? Is that the problem here?

Thanks for testing it @pablogs9

pablogs9, 2021-12-03:
I guess that it is related to your ROS 2 installation, can you test this inside a docker with a clean ROS 2 installation?

russkel, 2021-12-05:
I rm'ed `build` and `install` dirs and then ran build firmare/build agent again, and that seems to have fixed it. No idea what happened. Thanks for that.

closed 2021-12-05 by russkel