micro-ROS/rmw_microxrcedds #247: Size of memory pool item

ted-miller, 2022-04-22:
## Issue template

- Hardware description: Motoman robot and Ubuntu PC
- Version or commit hash: 51942f621a03a60f932f788a9c262e119dec3f99

#### Steps to reproduce the issue
1. Start agent

2. Start server:  (Modified rclc_examples) [rclc_examples(server).zip](https://github.com/micro-ROS/rmw_microxrcedds/files/8541927/rclc_examples.server.zip)
```
export RMW_IMPLEMENTATION=rmw_microxrcedds
cd microros_ws
source instal/local_setup.bash
ros2 run rclc_examples example_action_server
```

3. Start client [action_tester(client).zip](https://github.com/micro-ROS/rmw_microxrcedds/files/8541930/action_tester.client.zip)
```
cd microros_ws
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
source install/local_setup.bash
ros2 run action_tester fjt_test
```

#### Behavior
Any action-goal from the client that is greater than 896 bytes is not received by the server. The [deserialization fails in rmw_take_request](https://github.com/micro-ROS/rmw_microxrcedds/blob/fa98772887259cded1080a586ddd4b82a3228de2/rmw_microxrcedds_c/src/rmw_request.c#L124) because the buffer length is 896 bytes. 

I don't understand how the `rmw_uxrce_find_static_input_buffer_by_owner` function works. Where is it getting this memory block from? Why is it limited to 896?
Can I specify a custom memory block when initializing the action server? 
Can I enable dynamic allocation using my custom allocator? (I tried `DRMW_UXRCE_ALLOW_DYNAMIC_ALLOCATIONS`; it didn't help.)

I appreciate any advice you can provide. Right now, my FollowJointTrajectory server is limited to only a handful of trajectory points. (About 10, depending on the specific robot configuration.)

pablogs9, 2022-04-25:
[Buffer in RWM](https://github.com/micro-ROS/rmw_microxrcedds/blob/fa98772887259cded1080a586ddd4b82a3228de2/rmw_microxrcedds_c/src/rmw_microros_internal/types.h#L231) has the size determined by MTU and stream history, [check here](https://github.com/micro-ROS/rmw_microxrcedds/blob/fa98772887259cded1080a586ddd4b82a3228de2/rmw_microxrcedds_c/src/config.h.in#L50).

So, if you need to receive bigger messages just increase [MTU at the middleware layer](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/2195d43fff3052e3e426b8e678c682e5ffcc1cb5/CMakeLists.txt#L60).

Let us know if this works.

gavanderhoorn, 2022-04-25:
@pablogs9: in this particular case MTU is set to `2048`. The maximum sizes we seem to be able to receive are around the 1kB mark.

If I understand you correctly, we'd have to significantly increase MTU size to receive long(er) `JointTrajectoryAction` goals?

pablogs9, 2022-04-25:
The received data should fit in the buffers, so yes, if you need to receive longer `JointTrajectoryAction` goals you will need to increase the buffer size in order to be able to receive the data.

gavanderhoorn, 2022-04-25:
Just for my understanding: the `U` in `MTU` here then essentially refers to a *message* at the application level (ie: a ROS/DDS/XRCE-DDS message), and is not directly related to the MTU used by the transport(s)?

> if you need to receive longer `JointTrajectoryAction` goals you will need to increase the buffer size in order to be able to receive the data.

Would you be able to link to comments/documentation where we can learn about the relation between the different buffers, their sizes and the effects on data readers/writers?

pablogs9, 2022-04-25:
You can find some information [here](https://micro-xrce-dds.docs.eprosima.com/en/latest/optimization.html?highlight=mtu#stream-buffers) and [here](https://micro.ros.org/docs/tutorials/advanced/microxrcedds_rmw_configuration/)

As you said, MTU is the maximum transfer unit in bytes that will be used by the client and the agent during the XRCE session. Also, the client library defines the stream history, that is the number of slots that the client will use for fragmentation.

When the micro-ROS RMW uses the XRCE client library, it will use the configured MTU and will create [`RMW_UXRCE_STREAM_HISTORY_INPUT`](https://github.com/micro-ROS/rmw_microxrcedds/blob/fa98772887259cded1080a586ddd4b82a3228de2/rmw_microxrcedds_c/CMakeLists.txt#L88) slots of stream history. The former value should be the power of two.

So for example with the default configuration, your input buffer will have MTU * RMW_UXRCE_STREAM_HISTORY_INPUT = 512 * 4 = 2048 B. The maximum data size that can be received will be slightly less than this because of header overhead. If you want to receive messages larger than that, just increase the MTU or `RMW_UXRCE_STREAM_HISTORY_INPUT`.

pablogs9, 2022-04-25:
> if I understand you correctly, that would get us an 8 kB buffer?

That's it.

> Should that not be sufficient for quite long JTA goals?

Depends on the number of joints

pablogs9, 2022-04-25:
> (edit 2: does the quoted part of the text perhaps only apply to best-effort streams?)

Indeed, best-effort do not support fragmentation and the maximum size of the payload will be MTU less overhead.

> From the code referenced by @ted-miller and you (in https://github.com/micro-ROS/rmw_microxrcedds/issues/247#issuecomment-1108099611), it would seem in fact fragmentation does not allow receiving message(s) (fragments) larger than MTU*History. Is this because the resulting buffer is the "only space" available to store incoming fragments in, or is there some other more fundamental issue I'm overlooking?

That's correct.

> Perhaps I'm just confused by the term history: would it be correct to describe it as the depth of the queue holding incoming fragments waiting for reassembly?

That's correct.

> Edit 3: https://github.com/micro-ROS/rmw_microxrcedds/issues/247#issuecomment-1108099611 links to the setting for RMW_UXRCE_STREAM_HISTORY, but [Middleware Configuration](https://micro.ros.org/docs/tutorials/advanced/microxrcedds_rmw_configuration/) refers to [RMW_UXRCE_MAX_HISTORY](https://github.com/micro-ROS/rmw_microxrcedds/blob/fa98772887259cded1080a586ddd4b82a3228de2/rmw_microxrcedds_c/CMakeLists.txt#L48)?

[`RMW_UXRCE_STREAM_HISTORY`](https://github.com/micro-ROS/rmw_microxrcedds/blob/fa98772887259cded1080a586ddd4b82a3228de2/rmw_microxrcedds_c/CMakeLists.txt#L87-L91) is the XRCE layer history, the slots handled by the middleware layer.

[RMW_UXRCE_MAX_HISTORY](https://github.com/micro-ROS/rmw_microxrcedds/blob/fa98772887259cded1080a586ddd4b82a3228de2/rmw_microxrcedds_c/CMakeLists.txt#L48) is the RMW buffering slots, used for saving the topics received between the wait() and take() operations.

> Edit 4: would you have any tips/insights how to efficiently debug this? So far we've just noticed that "too large messages disappear". The Agent doesn't really output anything we can match to the buffer(s) being too small, nor the client as far as we can tell (but that's more difficult, as it's rather embedded).

If you need to debug the client-agent operation I would recommend building them in a Linux environment and debug normally with GDB in order to understand how it is behaving in terms of buffering.

In general:
- In best effort only < MTU can be sent and received.
- In reliable only < MTU * History can be sent and received.
- There is a special API that allows you to send more that MTU * History in a reliable stream as long as you provide a callback for flusing the buffers (normally running the session): [check here](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/2195d43fff3052e3e426b8e678c682e5ffcc1cb5/include/uxr/client/core/session/write_access.h#L130)
- There is a tweak for allow sending more than 64 kB (the limit imposed by the standard) in a write data operation: [check here](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/2195d43fff3052e3e426b8e678c682e5ffcc1cb5/CMakeLists.txt#L74)

gavanderhoorn, 2022-04-25:
> > (edit 2: does the quoted part of the text perhaps only apply to best-effort streams?)
> 
> Indeed, best-effort do not support fragmentation and the maximum size of the payload will be MTU less overhead.

It took me some time to figure this out. Perhaps the description under *Stream size* could be changed a bit to make it clearer?

> > Perhaps I'm just confused by the term history: would it be correct to describe it as the depth of the queue holding incoming fragments waiting for reassembly?
> 
> That's correct.

I haven't checked the XRCE-DDS standard docs: is fragmentation & reassembly described in terms referring to a "history" there, or is it RMW Micro-XRCE-DDS specific terminology? For some reason I assumed "history" to mean "message history", not "fragments" (but at some level each fragment *is* a message, so perhaps it's just me). I guess the use of *history* in 'regular' DDS QoS contexts also didn't align with what I thought I was reading here ('DDS samples' is about complete messages IIUC. The history here is more about message fragments).

Would updating the documentation to mention something like "depth of the buffer holding fragments waiting for reassembly" be an idea? It could perhaps make it easier to understand.

> > Edit 3: [#247 (comment)](https://github.com/micro-ROS/rmw_microxrcedds/issues/247#issuecomment-1108099611) links to the setting for RMW_UXRCE_STREAM_HISTORY, but [Middleware Configuration](https://micro.ros.org/docs/tutorials/advanced/microxrcedds_rmw_configuration/) refers to [RMW_UXRCE_MAX_HISTORY](https://github.com/micro-ROS/rmw_microxrcedds/blob/fa98772887259cded1080a586ddd4b82a3228de2/rmw_microxrcedds_c/CMakeLists.txt#L48)?
> 
> [`RMW_UXRCE_STREAM_HISTORY`](https://github.com/micro-ROS/rmw_microxrcedds/blob/fa98772887259cded1080a586ddd4b82a3228de2/rmw_microxrcedds_c/CMakeLists.txt#L87-L91) is the XRCE layer history, the slots handled by the middleware layer.

Ok. So the constant which defines the maximum nr of fragments which can be stored in the incoming msg buffer (with elements of (max) size `MTU`), correct?

> [RMW_UXRCE_MAX_HISTORY](https://github.com/micro-ROS/rmw_microxrcedds/blob/fa98772887259cded1080a586ddd4b82a3228de2/rmw_microxrcedds_c/CMakeLists.txt#L48) is the RMW buffering slots, used for saving the topics received between the wait() and take() operations.

Agreed, but [micro.ros.org/docs/tutorials/advanced/microxrcedds_rmw_configuration](https://micro.ros.org/docs/tutorials/advanced/microxrcedds_rmw_configuration/) seems to use `RMW_UXRCE_MAX_HISTORY` for calculating the size of the fragment buffer:

[image]

I'm probably misunderstanding something again though.

> If you need to debug the client-agent operation I would recommend building them in a Linux environment and debug normally with GDB in order to understand how it is behaving in terms of buffering.

Ok, thanks.

Would there be any signals/hooks the Client could register callbacks for to inform it of messages that didn't fit the receive buffers? Not to be able to 'save' them necessarily, but being aware of messages not making it up the stack could already be valuable. Especially while debugging.

pablogs9, 2022-04-26:
> It took me some time to figure this out. Perhaps the description under _Stream size_ could be changed a bit to make it clearer?

I'll take look.

> I haven't checked the XRCE-DDS standard docs: is fragmentation & reassembly described in terms referring to a "history" there, or is it RMW Micro-XRCE-DDS specific terminology? For some reason I [...]

The history here is a XRCE Stream history, this is that stores XRCE messages, that can contain Fragments. Please, refer to OMG XRCE standard.

> Ok. So the constant which defines the maximum nr of fragments which can be stored in the incoming msg buffer (with elements of (max) size `MTU`), correct?

Yes

> Agreed, but [micro.ros.org/docs/tutorials/advanced/microxrcedds_rmw_configuration](https://micro.ros.org/docs/tutorials/advanced/microxrcedds_rmw_configuration/) seems to use `RMW_UXRCE_MAX_HISTORY` [...]

I'm going to modify it: https://github.com/micro-ROS/micro-ROS.github.io/pull/369

> Ok, thanks.
>
> Would there be any signals/hooks the Client could register callbacks for to inform it of messages that didn't fit the receive buffers? Not to be able to 'save' them necessarily, but [...]

No, there aren't.

> Edit: [micro.ros.org/docs/tutorials/advanced/microxrcedds_rmw_configuration](https://micro.ros.org/docs/tutorials/advanced/microxrcedds_rmw_configuration/) also states:
> 
> > The size of the stream [..]
> 
> that confused me as well, as a stream -- by definition I believe -- doesn't really have a size or a length (or at least, not until you've received all of it and the connection has been closed).
> 
> Should that perhaps be _buffer_?
> 
> That terminology is used somewhat earlier on the same page (under _Micro XRCE-DDS_):
> 
> > On the one hand, best-effort streams could be interpreted as a single message queue. Therefore, the raw buffer is a single data buffer where only one message is popped/pushed.

It is named XRCE stream in the OMG XRCE standard because it is a stream of XRCE messages between the client and the agent.

ted-miller, 2022-04-26:
> So, if you need to receive bigger messages just increase [MTU at the middleware layer](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/2195d43fff3052e3e426b8e678c682e5ffcc1cb5/CMakeLists.txt#L60).
 Let us know if this works

Sorry for the delay in replying. My first test of increasing these values was not successful. However, I got pulled into other issues before I could debug any further. I will be back on this later during the week.

ted-miller, 2022-04-28:
> [Buffer in RWM](https://github.com/micro-ROS/rmw_microxrcedds/blob/fa98772887259cded1080a586ddd4b82a3228de2/rmw_microxrcedds_c/src/rmw_microros_internal/types.h#L231) has the size determined by MTU [...]
 So, if you need to receive bigger messages just increase [MTU at the middleware layer](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/2195d43fff3052e3e426b8e678c682e5ffcc1cb5/CMakeLists.txt#L60).
 Let us know if this works.

This did not solve the problem that we're experiencing. But, I have confirmed that the problem is not due to the size of this buffer. So, I am closing this issue while I investigate the true source of the problem.

Thank you for the advice about increasing the history and mtu sizes. We will definitely take advantage of this going forward.

ted-miller, 2022-04-29:
Continued at: https://github.com/micro-ROS/rosidl_typesupport_microxrcedds/issues/48