micro-ROS/rmw_microxrcedds #259: rmw_uros_ping_agent fails after update on Teensy 4.0

v-lopez, 2022-06-02:
**Describe the bug**
No matter which values I have on the `rmw_uros_ping_agent` timeout and retries, they always fail. But I can publish messages normally.

**To Reproduce**
Steps to reproduce the behaviour:
1.  Update Arduino library using  `docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:galactic teensy4`
2. Use https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino
3. Upload
4. Run the micro ros agent with `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -v`
5. The board continuously reconnect because ping fails

**Expected behaviour**
The code does not reconnect unless the board is unplugged.

**System information (please complete the following information):**
 - OS: Ubuntu 20.04
 - ROS 2 Galactic
 - Version Latest

**Additional context**
Using this packaged library downloaded a few months ago works as intended. 
[micro_ros_arduino-2.0.2-galactic.tar.gz](https://github.com/micro-ROS/rmw_microxrcedds/files/8823614/micro_ros_arduino-2.0.2-galactic.tar.gz)

I cannot use the old library as I need to pack custom messages, so I need to use the docker command, which forces me to use the last version which is broken.

pablogs9, 2022-06-02:
Which version of the micro-ROS Agent are you using?

v-lopez, 2022-06-02:
I am using master of https://github.com/eProsima/Micro-XRCE-DDS-Agent
And 2.0.1 of micro-ROS-Agent (https://github.com/micro-ROS/micro-ROS-Agent/commit/52b35ef2eb9ea24bced342f6fc27f881ae5600cb)
I can update to 2.0.2, but the changes seem unrelated to this.

pablogs9, 2022-06-02:
Why are you handling [Micro-XRCE-DDS-Agent](https://github.com/eProsima/Micro-XRCE-DDS-Agent) manually? It should be downloaded automatically by the micro-ROS Agent build process.

v-lopez, 2022-06-02:
Ok I believe my problem was due to using `master` of https://github.com/eProsima/Micro-XRCE-DDS-Agent instead of `ros2`.

There's  2/3 commit difference but it's interfering with the ping. 

I know what I am doing is outside of the standard use, but those branch names can be confusing.

pablogs9, 2022-06-03:
`ros2` is a branch that follows `develop` in Micro XRCE-DDS Agent. There are modifications in ping behavior in the latest versions of `develop` and they have not been released to `master` yet. Let us know if using `ros2` branch works as expected.

v-lopez, 2022-06-03:
Yes, as soon as I switched to `ros2` I stopped having the reconnection problem, both on the example code I linked above as well as on my custom code.

pablogs9, 2022-06-03:
Nice, closing. thanks for the report

closed 2022-06-03 by pablogs9