micro-ROS/rmw_microxrcedds #266: I wanted to use MicroROS without any custom transport or RMW. Is it possible to bypass the RMW and use only the publish-subscribe?

NagaHimanshu, 2022-08-25:


pablogs9, 2022-08-25:
Please elaborate on your issue, requirements and goals.

closed 2022-08-25 by pablogs9

NagaHimanshu, 2022-08-25:
I wanted to develop a basic publish-subscribe model using MicroROS without any UART/TCP/UDP. I'd like to open two threads using FreeRTOS, one thread for publish and another thread for subscribe. For this I created a dummy transport which just fills up memory. I attached a piece of code below. So, is it possible to having this kind of scenario?

[New Text Document.txt](https://github.com/micro-ROS/rmw_microxrcedds/files/9424006/New.Text.Document.txt)

pablogs9, 2022-08-25:
You can explore the shared memory in Micro XRCE-DDS Client: https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/6f4edd25c0984fd30bf3bf84e2d1d33af03f24a1/CMakeLists.txt#L68

NagaHimanshu, 2022-08-25:
Thank you so much for the suggestion. It is working as per my requirement. Thanks again.