micro-ROS/rmw_microxrcedds #297: High CPU usage with int32_publisher

yashi, 2023-05-31:
**Describe the bug**
Having only timers on an executor makes `rmw_wait()` to spin without blocking.  This causes high cpu usage and it's a problem on a RTOS with threading support.

**To Reproduce**
Just build `int32_publisher` from [micro-ROS-demos](https://github.com/micro-ROS/micro-ROS-demos).

**Expected behaviour**
`rclc_executor_spin()` should always be a blocking call without any service, client, subscription, or timer.

**System information (please complete the following information):**
 - OS: Zephyr (but it shouldn't matter)
 - ROS 2: Humble (but it shouldn't matter)
 - Version: humble.

**Additional context**
https://micro-ros.slack.com/archives/CNC141G0N/p1684752941054099

Acuadros95, 2023-07-31:
This is resolved by https://github.com/micro-ROS/rmw_microxrcedds/pull/305, please close when merged

yashi, 2023-08-01:
Thanks for you PR.

BTW, if you have "Resolves #297" in the description of #305, this will be closed when merged.
https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue#linking-a-pull-request-to-an-issue-using-a-keyword