micro-ROS/rmw_microxrcedds #306: rmw_uros_ping_agent always adds 100 ms delay

mvanlobensels, 2024-03-27:
**Describe the bug**
The `rmw_uros_ping_agent` function adds 100 ms of delay to the function's `timeout_ms` input argument when the Micro-ROS Agent is not found or not connected.

For example, when measuring the time it takes to evaluate `rmw_uros_ping_agent(10, 1)`, the result is 110 ms.
Similarly, `rmw_uros_ping_agent(100, 1)` takes 200 ms.
`rmw_uros_ping_agent(10, 2)` results in 210 ms.

**To Reproduce**
I am using the [micro-ros_reconnection_example](https://github.com/micro-ROS/micro_ros_arduino/blob/iron/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino) to reconnect to the Agent in the case that a connection is lost, or when my microcontroller boots before the PC which runs the Micro-ROS Agent boots.

```c
switch(agent_state)
{
    case WAITING_AGENT:
        time_start = uxr_millis();        
        agent_state = (RMW_RET_OK == rmw_uros_ping_agent(10, 1)) ? AGENT_AVAILABLE : WAITING_AGENT;
        time_end = uxr_millis() - time_start;
        break;
    case AGENT_AVAILABLE:
        ...
```

**Expected behaviour**
I expect the delay to be equal to `timeout_ms`. We are controlling electric motors with the microcontroller, so adding a 100 ms delay to our system makes our robot uncontrollable.

**System information:**
- Microcontroller: Renesas RA6M5
 - OS: Pop OS 22.04
 - ROS 2 Humble

pablogs9, 2024-04-01:
Sorry but as far as we do not support Pop OS 22.04 for Renesas RA6M5 I'm not sure about the behaviour of your system. Could you confirm that it behaves the same with a supported configuration?

mvanlobensels, 2024-04-04:
Pop OS is the practically the same as Ubuntu, with drivers pre-installed and customized GNOME features. I find it hard to believe that that could be the issue.
https://support.system76.com/articles/difference-between-pop-ubuntu/

So this additional 100 ms is not something that is implemented by design?

mvanlobensels, 2024-04-09:
I am using Serial USB to communicate between the Agent and the microcontroller

pablogs9, 2024-04-09:
Could you check if changing the timeout here: https://github.com/micro-ROS/micro_ros_renesas2estudio_component/blob/a99ccd815ebe6de4b53b02189cedd36bb1847c7d/extra_sources/microros_transports/usb_transport.c#L136 solves the issue?

Not sure right now if you can set it to

```cpp
return handle_usb(USB_WRITE, (uint8_t *) buf, len, 0);
```

but maybe something less than WRITE_TIMEOUT (that is 100 ms).

mvanlobensels, 2024-04-11:
Thank you. This was indeed the cause of the 100 ms delay.

closed 2024-04-11 by mvanlobensels