micro-ROS/rmw_microxrcedds #312: Potential Issue with rclc_action_server_init_default

DannyTravis1, 2025-01-13:
## Issue template

- Hardware description: Custom board using STM32F412VG processor
- RTOS: FreeRTOS plus posix
- Installation type: custom static micro-ros library
- Version or commit hash: humble

#### Steps to reproduce the issue
I integrated a custom micro-ros static library onto my project using USB as the custom transport. It has 3 publishers and 2 action servers. Whenever i get to rclc_action_server_init_default in my init function (that was a mouthful, sorry for the run on sentence), I can see the services initiating on the MCU side and I see the agent side responding (See agent log in additional information). However, when I get to the Feedback Publisher, in paticular when I get to uxr_run_session_until_all_status I get a timeout and then the whole thing fails
[image]

my init function looks like this:
```
/* **************************** *
 * Action servers init function 
 * **************************** */
bool init_prop_action(void){
	uros_node.prop_srv.act_name = "/prop_act";
	uros_node.prop_srv.rcl_ret = RCL_RET_OK;
	uros_node.prop_srv.rcl_ret  =
	rclc_action_server_init_default(&uros_node.prop_srv.action_server,
	&uros_node.node,
	&uros_node.support,
	ROSIDL_GET_ACTION_TYPE_SUPPORT(plato_interfaces,PropControl),
	"/property"
	);

	
	if(uros_node.prop_srv.rcl_ret == RCL_RET_OK){
		uros_node.prop_srv.rcl_ret =
		rclc_executor_add_action_server(&uros_node.exe_actions,
		&uros_node.prop_srv.action_server,
		6,
		&uros_node.prop_srv.goal_req_msg,
		sizeof(plato_interfaces__action__PropControl_SendGoal_Request),
		prop_handle_goal,
		prop_handle_cancel,
		(void*)&uros_node.prop_srv.action_server);
	} else {
		Debug_printf("TL: err prop server init RCL Ret: %d\n\r", uros_node.prop_srv.rcl_ret);
		return false;
	}
	if (uros_node.prop_srv.rcl_ret == RCL_RET_OK){
		return true;
	}
	Debug_printf("TL: err prop server exec RCL Ret: %d\n\r", uros_node.prop_srv.rcl_ret);
	return false;

}
```
the rcl ret is 1

#### Expected behavior
When the agent registers the feedback publisher and topic it should move on to the status publisher

#### Actual behavior
program fails at the macro PUBLISHER_INIT(feedback) in rcl_action_server_init();

#### Additional information
[agent_log.txt](https://github.com/user-attachments/files/18402356/agent_log.txt)

DannyTravis1, 2025-01-14:
Just to add visibility, the issue happens when I create this publisher (I don't see it when I create a stand alone publisher).
The publisher init function (rmw_create_publisher) calls run_xrce_session:
[image]
And then in the run function the application gets stuck at uxr_run_session_until_all_status:
[image]
I'm still trying to figure out where uxr_run_session_until_all_status is in the code, but once I do I'll see if I can add any additional info that might help.

Samthnalyst1234, 2025-01-17:
I can see that the issue is occurring in the XRCE-DDS layer during publisher creation, specifically at the uxr_run_session_until_all_status call. Let's try this:
First, let's modify the timeout handling in your action server initialization. Since we now know the failure point is at session status verification, we need to:
`bool init_prop_action(void) {
    // First, ensure session is alive
    if (!rmw_uros_ping_agent(100, 2)) {  // Try 2 times with 100ms timeout
        Debug_printf("TL: Agent connection check failed\n\r");
        return false;
    }

    // Set longer timeout for XRCE-DDS operations
    uxrSessionTimeout timeout = {1000, 1000000}; // 1 second timeout
    rmw_context_impl_t* context = uros_node.node.context->impl;
    uxr_set_timeout(&context->session, &timeout);

    // Your existing action server initialization code
    uros_node.prop_srv.act_name = "/prop_act";
    uros_node.prop_srv.rcl_ret = rclc_action_server_init_default(
        &uros_node.prop_srv.action_server,
        &uros_node.node,
        &uros_node.support,
        ROSIDL_GET_ACTION_TYPE_SUPPORT(plato_interfaces, PropControl),
        "/property"
    );

    // Add debug prints
    if (uros_node.prop_srv.rcl_ret != RCL_RET_OK) {
        Debug_printf("TL: Action server init failed at step %d with code: %d\n\r", 
                    uros_node.prop_srv.rcl_ret);
        return false;
    }

    // Rest of your initialization code...
}`

Samthnalyst1234, 2025-01-17:
If you're using USB transport, add these transport configurations.......

// Add before action server initialization
rmw_uros_set_custom_transport(
    true,                          // Reliable stream
    NULL,                          // Custom args
    usb_transport_open,            
    usb_transport_close,
    usb_transport_write,
    usb_transport_read
);

Also, add these reliability settings:

rcl_action_server_options_t server_ops = rcl_action_server_get_default_options();
server_ops.goal_service_qos = rmw_qos_profile_services_default;
server_ops.result_service_qos = rmw_qos_profile_services_default;
server_ops.cancel_service_qos = rmw_qos_profile_services_default;
server_ops.feedback_topic_qos = rmw_qos_profile_default;
server_ops.status_topic_qos = rmw_qos_profile_default;

Samthnalyst1234, 2025-01-17:
The SPI errors in your logs suggest potential transport issues. If you're using SPI transport, ensure your transport configuration includes:
#define TRANSPORT_MTU 512
#define TRANSPORT_OVERHEAD 8

// Adjust these based on your STM32F412VG configuration
static size_t transport_write(void* buffer, size_t length, void* args) {
    (void)args;
    size_t written = 0;
    while (written < length) {
        size_t chunk = MIN(TRANSPORT_MTU - TRANSPORT_OVERHEAD, length - written);
        // Your SPI write implementation
        written += chunk;
    }
    return written;
}

Samthnalyst1234, 2025-01-17:
Finally, add error reporting for the session status:

if (!uxr_run_session_until_all_status(&session, timeout, request, &status, 1)) {
    uint8_t errcode = uxr_get_status(&session, request);
    Debug_printf("TL: Session status error: %d\n\r", errcode);
    return false;
}

Samthnalyst1234, 2025-01-17:
The main issues appear to be:

Timeout during session status verification
Potential transport layer issues (SPI errors)
Possible QoS mismatch between publisher and subscriber

Can you try implementing these changes and let me know if the error persists? Also, could you confirm:

Which transport you're using (USB or SPI)?
The MTU size configured for your transport?
Whether regular publishers work without the action server?

waqasjavaid8, 2025-01-17:
Hi @DannyTravis1 

I have seen your project's issue in details. The problem of things fail due to timeout is because of a misconfiguration in the feedback publisher's initialization. I have checked your images but I don't have access the complete code. So that I can modify it for you.

I am waqas come from your upwork post. Please contact me on my email or whatsapp. So that I can solve your issue in 3 to 4 hours.

Email: waqasjavaid744@gmail.com
Whatsapp: +923345057878

Thanks
waqas

DannyTravis1, 2025-01-24:
Thanks for getting back too me! @lcbw   is completely correct, I'm making it through the uxr_flash_output_streams (which does the flush on the output stream). Im seeing the Agent terminal get the messages, but the only thing i see the agent sending is the heartbeat message.  Where the red arrow is is when I get past the flush and then hit the next function, listen_message_reliably

[image]

As far as I can tell Im not receiving a response from the agent, but I can see them reading and writing. Is there something special I need to do on the agent side? Im running the humble docker container

DannyTravis1, 2025-01-24:
Also, pretty sure I'm reading, I set this break during the listen function and I can see the ringbuffer filling up

[image]

DannyTravis1, 2025-01-27:
##UPDATE
Well this is not cool. I added a debug print to make sure that I'm reading and it fixed the issue:

[image]

My read function is just grabbing from a ringbuffer that's filled byte by byte from a USB ISR routine. I'm going to see if maybe there's something wrong with the USB driver, because a print statement shouldn't do anything.

marip8, 2025-01-28:
@gavanderhoorn do you have any insight into what might be causing this issue? @lcbw has been helping diagnose up to this point but we don't have too much experience with RCL C development

waqasjavaid8, 2025-01-28:
Hi @marip8 @DannyTravis1  I can help you in your projects.

Please connect me 

Email: waqasjavaid744@gmail.com
Whatsapp: +923345057878

Thanks
waqas

DannyTravis1, 2025-01-28:
AND FIXED!
Turns out that when you run a non blocking transport (like what I'm doing) you can return from the read function BEFORE the timeout.
The reason why the print statement fixed it is because it caused a delay before the function returns! However, the return has to happen AT the timeout.
i.e. adding a delay that equals the timeout fixed if: 

[image]

closed 2025-02-28 by DannyTravis1