micro-ROS/rmw_microxrcedds #314: 5 digit port numbers are truncated

jimmy-mcelwain, 2025-02-27:
Context https://github.com/Yaskawa-Global/motoros2/issues/376

**Describe the bug**
If a 5 digit port number is specified, the last digit gets cut off. 

**To Reproduce**
Specify a 5 digit port number. If I choose 16510, then the "real" port number ends up being 1651. 

**Expected behaviour**
If I specify a 5 digit port number, the last digit should not be truncated. It should alarm for 0 or 6+ digits. 

**System information (please complete the following information):**
 - Found on a YRC1000. Client PC is running micro-ros agent. 
 - Found on ROS2 Jazzy
 - 5.01

**Additional context**

The problem lies here:

https://github.com/micro-ROS/rmw_microxrcedds/blob/554445fe739e4c5bcb27753ffbe21e03bb2c3c1f/rmw_microxrcedds_c/src/rmw_microros/init_options.c#L100-L105

With the maximum length argument in `snprintf` being `MAX_PORT_LENGTH`, that means that at most `MAX_PORT_LENGTH - 1` characters will be copied into the `agent_port` array, since it needs room to put the terminating `\0`.

pablogs9, 2025-02-27:
Good catch, thanks.

https://github.com/micro-ROS/rmw_microxrcedds/pull/315

pablogs9, 2025-03-06:
Closing as #315 merged

closed 2025-03-06 by pablogs9