micro-ROS/rmw_microxrcedds #322: Feature request: runtime switching of transport channel

shvass, 2025-03-12:
## Support for runtime switching of transport (UDP / custom, etc )

Current implementation requires compile time definition for transport type. Adding  support for runtime
switching will open possibility of applications such as a backup channel in case main channel fails.

### possible implementation
- add a another transport "multimode" apart from UDP, custom, etc which defines multimode implementation.  Thus only users with "multimode" bear the overhead of binary size and execution. 

#### multimode
- [rmw_uxrce_transport_params_t](https://github.com/micro-ROS/rmw_microxrcedds/blob/jazzy/rmw_microxrcedds_c/include/rmw_microros/rmw_microros.h#L55) contains parameters for all possible types. Nothing excluded by macros.

- Add a static global variable "RMW_UXRCE_ACTIVE_TRANSPORT" to determine  currently active transport and make transport call as per currently active transport.

+ many more

Does this sound like a good idea ?

pablogs9, 2025-03-12:
You always can use `RMW_UXRCE_TRANSPORT_CUSTOM` and use the API to change transports:

https://github.com/micro-ROS/rmw_microxrcedds/blob/jazzy/rmw_microxrcedds_c/include/rmw_microros/custom_transport.h#L54

I've never done that and there might be some implication with regards to RMW initialization or deinitialization. 
But at least in theory, micro-ROS RMW custom transport are run-time configured.

shvass, 2025-03-12:
I'll definitely try out the "custom transport."

    Wouldn't it be great if multimode were a core feature of micro-ROS? What are your thoughts on this?
    Would a PR for multimode be considered?